ArduPilot/MissionPlanner #1300: Log download time out

Fossdal, 2016-06-25:
MissionPlanner 1.3.38.1 build 1.1.6020.16327
Frame: QUAD
ArduCopter V3.2.1 (36b405fb)
APM 2.6 clone

I am unable to download logs connected by telemetri link 
Under Flight Data/Data Flash Logs/ Download DataFlash Log Via Mavlink
i get a list with logs (26 logs)
i choose a log and press "Download Selected Logs" ti starts to download and i see the progressbar increase and counting up
this is from the output window

Getting list of log files...
Found 26 log files, note: item sizes are just an estimate.
Fetching log file 01-01-1970 00:00:00 ...
Error in log 19: Timeout on read - GetLog

Fossdal, 2016-06-25:
I have downloaded the source and modified line 3792 in MAVLinkInterface and put 3000 insted of 1000
then i don't get timeout, and am able to download logs

BUT i get at lot of CRC errors
2016-06-25 14:17:33,108  INFO MissionPlanner.MAVLinkInterface - Mavlink Bad Packet (crc fail) len 105 crc 11421 vs 56579 pkno 120 LOG_DATA (C:\DEV\MissionPlanner-master\Mavlink\MAVLinkInterface.cs:3125) [Log download single thread]

meee1, 2016-06-25:
are you downloading over usb? or radio?

Fossdal, 2016-06-26:
over radio

meee1, 2016-06-28:
I would suggest using usb. you will always get packet loss over radio. also it will be 3 times quicker over usb.

meee1, 2016-06-28:
closing, as fixed with code, and suggestion

closed 2016-06-28 by meee1