ArduPilot/MissionPlanner #1308: Polygon fence - possible improvements to interface

rmackay9, 2016-07-12:
Some suggestions on enhancing the GeoFence setup page to make it easier to setup the polygon fence.

[image]

lvale, 2016-07-12:
@rmackay9 you've missed the load/save to disk of the fence file.

lvale, 2016-07-12:
Also, as we've discussed perhaps also adding a circular fence in different color (if enabled) with a single radial point that could be moved to set the circular fence radius, from Home Location.

rmackay9, 2016-11-15:
It came up again on the AC3.4 forum that people would like to be able to more easily clear the fence.
http://discuss.ardupilot.org/t/clearing-a-polygon-fence/12691/11

Tobinejumi, 2016-11-20:
Also, if it's possible - ability to set up Ch6 (or other) for "Fence disable/enable" command.