ArduPilot/MissionPlanner #1309: Request: HUD indication that Quadplane vertical motors are operating.

iskess, 2016-07-12:
I would like to be able to see real time if/when Q_ASSIST is activated during a flight. I can use the HUD User Items to list CH3 Throttle, but we need a similar Throttle indication for the average of the copter motors. This would also make graphing easier.

meee1, 2016-07-23:
do you have a tlog with inflight data? and being in q_assist?

iskess, 2016-07-23:
Thanks Michael. In the middle of the flight, about 60%, I slow down and let the wind push the plane backward with Q_Assist active, then transition back to forward flight. 
[2016-07-15 11-28-30.zip](https://github.com/ArduPilot/MissionPlanner/files/379445/2016-07-15.11-28-30.zip)

iskess, 2017-07-09:
I did a lot of Quadplane recently and I'm really missing this feature. I need an easy way to identify when the throttle setting of the quad motors.