ArduPilot/MissionPlanner #1311: FLIR Vue for camerasBuiltin.xml

0n3-70uch, 2016-07-13:
Hello,

the FLIR Vue parameters are wrong in the "camerasBuiltin.xml".
Please remove the FLIR parameter and add the correct.

`<Camera>
    <name>FLIR Vue 336 6,8mm</name>
    <flen>6.8</flen>
    <imgh>256</imgh>
    <imgw>336</imgw>
    <senh>4.352</senh>
    <senw>5.712</senw>
  </Camera>
  <Camera>
    <name>FLIR Vue 336 9mm</name>
    <flen>9</flen>
    <imgh>256</imgh>
    <imgw>336</imgw>
    <senh>4.352</senh>
    <senw>5.712</senw>
  </Camera>
  <Camera>
    <name>FLIR Vue 336 13mm</name>
    <flen>13</flen>
    <imgh>256</imgh>
    <imgw>336</imgw>
    <senh>4.352</senh>
    <senw>5.712</senw>
  </Camera>
  <Camera>
    <name>FLIR Vue 640 9mm</name>
    <flen>9</flen>
    <imgh>512</imgh>
    <imgw>640</imgw>
    <senh>8.704</senh>
    <senw>10.88</senw>
  </Camera>
  <Camera>
    <name>FLIR Vue 640 13mm</name>
    <flen>13</flen>
    <imgh>512</imgh>
    <imgw>640</imgw>
    <senh>8.704</senh>
    <senw>10.88</senw>
  </Camera>
  <Camera>
    <name>FLIR Vue 640 19mm</name>
    <flen>19</flen>
    <imgh>512</imgh>
    <imgw>640</imgw>
    <senh>8.704</senh>
    <senw>10.88</senw>
  </Camera>`

Greetings
Fabian

meee1, 2016-07-23:
what should they be?

0n3-70uch, 2016-07-23:
Oh sorry. Here are the specs:

`<Camera><name>FLIR Vue 336 6,8mm</name><flen>6.8</flen><imgh>256</imgh><imgw>336</imgw><senh>4.352</senh><senw>5.712</senw></Camera><Camera><name>FLIR Vue 336 9mm</name><flen>9</flen><imgh>256</imgh><imgw>336</imgw><senh>4.352</senh><senw>5.712</senw></Camera><Camera><name>FLIR Vue 336 13mm</name><flen>13</flen><imgh>256</imgh><imgw>336</imgw><senh>4.352</senh><senw>5.712</senw></Camera><Camera><name>FLIR Vue 640 9mm</name><flen>9</flen><imgh>512</imgh><imgw>640</imgw><senh>8.704</senh><senw>10.88</senw></Camera><Camera><name>FLIR Vue 640 13mm</name><flen>13</flen><imgh>512</imgh><imgw>640</imgw><senh>8.704</senh><senw>10.88</senw></Camera><Camera><name>FLIR Vue 640 19mm</name><flen>19</flen><imgh>512</imgh><imgw>640</imgw><senh>8.704</senh><senw>10.88</senw></Camera>`

meee1, 2016-07-27:
done thanks.

closed 2016-07-27 by meee1