ArduPilot/MissionPlanner #1313: Download All Logs rewriting old files

vfp1, 2016-07-18:
Hi, 
thank you for this amazing piece of software. In earlier versions when my missions were completed I usually followed the practice of downloading all the logs via "Download DataFlash Log Via Mavlink", and then selected "Download All Logs". Usually the files were never overwritten and they were stored in the log folder in Mission Planner without being modified. Since the APM consistently makes space for new data, old logs are being progressively erased as new logs are coming in. Thus, when I download the all the logs, the old ones are overwritten with the ones that are missing some data that was erased to make space for new logs. So I found that I have some logs with half of the data missing. 

I am not sure if this is something new or some bad procedure that I am following. At the moment I just try to remember which were the newest logs and try to download those, but for security reasons it will be nice to download all logs every time. 

Best,

meee1, 2016-07-23:
what version of MP are you running?

vfp1, 2016-07-24:
Hi, thanks for the reply. I am using Mission Planner version "1.3.38 build 1.1.5983.12141" in Windows 10.

meee1, 2016-07-24:
please update to the latest
issue has ben solved
https://github.com/ArduPilot/MissionPlanner/blob/master/ChangeLog.txt#L108

vfp1, 2016-07-24:
That's great then, thank you very much!

closed 2016-07-24 by vfp1