ArduPilot/MissionPlanner #1315: Dataflash log auto anaysis appears to be broken in 1.3.39

mtbsteve, 2016-07-20:
I just upgraded MP to 1.3.39. I just installed the update to MP 1.3.39 and realized that the Dataflash auto analysis seems to be broken.
All my .bin files from AC 3.2.1 as well as AC 3.3 and Solo I could auto-analyze in the previous versions of MP result in a "Failed to start LogAnalyzer" followed by a "bad input file" message with this release. 
Analyzing with the  interactive "Review a log" function works fine however.

meee1, 2016-07-21:
please provide more info.

OS and version. etc

mtbsteve, 2016-07-21:
Sure. 
I am running Windows 10 on a Laptop PC equipped with a Cerleron 32bit CPU with 4GB RAM and about 40GB of free disk space. Error is reproducible with AC 3.2.1 or AC 3.3 .bin files of any size.

meee1, 2016-07-27:
ive made some mods to mp beta. now has a 32bit and 64 bit version.

mtbsteve, 2016-07-27:
+1 thanks!

Fossdal, 2016-08-28:
I still get this Error 
Downloaded source from github today 28/08-2016
when it downloaded the logAnalyzer it downloaded the 64bit version
OS Windows 10 64bit, 
8GB ram

magicrub, 2016-10-05:
Same as https://github.com/ArduPilot/MissionPlanner/issues/1370

Patnetpatnet, 2016-10-05:
Same issue. Deleting software and re-installing with an earlier ver. fixes but throws error when uploading params from FC.

Fossdal, 2016-10-07:
Version 1.3.41.3 Build 1.1.6122.35387 now works
Windows 10 64bit