ArduPilot/MissionPlanner #1316: Load custom firmware, Reading Hex File issue

olliw42, 2016-07-24:
hey

I'm having strange issues with flashing custom files using MP, and am on this since two long days now without progress. 

I'm on Win7 and am using PX4 Eclipse of the PX4 Toolchain v14 to compile modified ArduCopter3.4-rc1 code (make px4-v2), and flash it using MP's Install Firmware->Load custom firmware option.  I tested MP 1.3.38 and 1.3.39. At least two issues:

(1) I select the firmware .px4 file, but very often the following "Reading Hex File" takes literally ages ... I have difficulties to understand that given that the .px4 is on local disk. So, that seems to indicate another problem. What could that be?

(2) Often it mistakenly takes a new .px4 and believes it is identical to the currently uploaded code. Example: I modify the THISFIRMWARE string by counting up a number, to double check the flashed firmware. I have e.g. firmware "blabla 002-10" flashed, that is I connect to the board and check the Messages and see "blabla 002-10". I now change that in Eclipse to e.g. "blabla 002-11", compile, and start the upload. MP connects (if it passes the long Reading Hex File issue), but detects it's identical to the flashed one. This can happen even if I explicitly have deleted from disk the old .px4 file before doing the compile. Often I can (have to) repeat that several times, and sometimes might even have to flash another version altogether, to make MP flash the real file.

I have no idea what is going on, it somehow seems as if MP can confuse itself about which .px4 data to flash.

thx

meee1, 2016-07-24:
could you please provide your .px4 file?

olliw42, 2016-07-25:
I will, this evening (if by then you still think its needed)
the night yielded some more "insights":

on (1): I might not have expressed myself well. What I mean is: I believe that while MP displays "Reading Hex File" it not only opens and reads the file but in addition does all sorts of other things, and that it are these other things which make it to get stuck for long. So, my question essentially was, what else is it trying to do in all that time? Knowing that might help me better find the issue.

on (2): I thought (1) and (2) might point to the same underlying issue, but it now seems they are not, and (2) might not be a MP issue at all, but an issue of the PX4 Toolchain. I nevertheless report the new observation here, since I really don't know better (sorry). The new news:

I extended the ac3.4-rc1 code to output some debug messages on a serial, which I read with a terminal. I made it to output exactly the THISFIRMWARE define. So, I change this define, compile, flash, then look at the message in the terminal, connect with MP to the pix board and compare with the message displayed by MP. They don't always match. Please see this screenshot here: http://www.olliw.eu/drop/ac-mp-version-issue.jpg. 

Note, this is even so then I do quite substantial changes to the code (i.e. not only change the version.h). It is however not always so. Frequently, for reasons I could not figure out, the MP message is getting changed. But frequently, it is not. A px4-v2-clean however does seem to reliably correct it, not surprisingly. 

Since I have that serial debug output in the code, MP never again has refused to flash claiming identical codes (that is even in cases as shown in the screenshot). I could not yet better pin down why it happend before.

As said I don't know on which end the issue is, I don't understand these parts well, maybe it is also related to the issues why a new Px4 toolchain is in the making. It's wired though.

meee1, 2016-07-27:
for 1. i did this https://github.com/ArduPilot/MissionPlanner/commit/d1e104b27df23008c817ff710391c573ea181c24

just added some more steps in the progress.

meee1, 2016-07-27:
for upload, mp calcs a checksum of the file, and then compares it to the checksum returned by the board.
i would guess it is indeed a toolchain issue. might need to clean between builds

olliw42, 2016-07-30:
MANY thanks, I will test as soon as there is a new MP version

on (2), yes, seems a toolchain thing, just wanted to hear your opinion 
a clean does the trick, but cleans are very time consuming on Win, hopefully these kind of compile inconsistencies will be gone with toolchain v1.5. 

THX again!

closed 2016-08-16 by meee1