ArduPilot/MissionPlanner #1317: include RC12 - RC14 in "Servo" tab

iskess, 2016-07-25:
Why does the Servo tab only include channels RC5-RC11? I'm wanting to activate servos on channel RC13. Has this just not been updated since the APM 2.6, or is there some other limitation?

[image]

iskess, 2016-07-25:
Also, the Camera Setup page doesn't allow any channels above RC11. Not a really an issue because I can set up in the full parameters, but thought I would mention it.

I can't think of any work around for manually activating a servo, and still having it available for DO_SET_SERVO. To my knowledge there is no way to use my RC transmitter to move a servo once allowing the APM to control it.  This Servo control tab is the only way I know.

meee1, 2016-08-16:
added

closed 2016-08-16 by meee1