ArduPilot/MissionPlanner #1318: camera footprint and overlap

StephBMG, 2016-07-25:
In survey (grid) and Flight Data (camera overlap):
The footprint displayed is not equal to the footprint calculated
the error increase when the focal length decrease
example in attached files for a 10mm lens on full frame camera at 1150 ft with 0% overlap and 0% sidelap, the footprint should cover the area

[image]
[image]
[image]

meee1, 2016-07-25:
what is your home location and altitude set to? in flightplanner?

StephBMG, 2016-07-26:
Hi, lat -15,4380027549346 Long 49,6896815299988 Alt 0

StephBMG, 2016-07-26:
Hi,
lat -15,4380027549346 Long 49,6896815299988 Alt 0

in attached file the grid file

thanks

Stéphane

FLYBYME, 2016-08-05:
I have the same issue.

meee1, 2016-08-05:
the latest beta fix's this, I'm still working on the long term solution though.

FLYBYME, 2016-08-05:
OK I just installed 1.39 or something. What version has the fix?

meee1, 2016-08-16:
MP beta will have the fix.
help > update beta