ArduPilot/MissionPlanner #1321: When setting param BLAH_ENABLE (or similar) then refresh params

magicrub, 2016-08-02:
When setting many of the XXXXXX_ENABLE params the latest versions of ArduPilot will hide params. For example, if you set ADSB_ENABLE=0 then all other ADSB_xxxxx params disapear. A fetch will only show ADSB_ENABLE=0. When setting ADSB_ENABLE=1 then the autopilot makes the ADSB_xxxx params available but you won't know it until you refresh params.

TODO:
When reading all params, make a note of the param count. When writing a param you should check the param response packet and see if the param count value has changed. If so, then a full refresh should be performed.

meee1, 2016-08-04:
@magicrub I like the TODO part, you have confirmed it does respond like this?

magicrub, 2016-08-04:
yup. and I just pushed the changes today to make it work for all libraries

meee1, 2016-08-04:
@magicrub  :)

magicrub, 2016-08-07:
don't forget, that if you write a param that is in a hidden library then it fail. So if you're writing BLAH_PARM1 while BLAH_ENABLE=0 it will fail. So when writing *_ENABLE=0 then you have to write those lib's params first, then the ENABLE=0. Opposite goes for *_ENABLE=1 you should write enable first then the rest of the params.

This sure made the process more complicated! :(