ArduPilot/MissionPlanner #1329: Add new Flight Mode

eduherminio, 2016-08-11:
Once the new flight mode is implemented in the code, according to other closes issues it's necesary to modify ParameterMetaData.xml (in MP directory) to include the flight mode among the 'official' ones.
Now ParameterMetaData.xml doesn't seem to exist anymore, and it seems to have been replaced with ParameterFactMetaData.xml and ParameterMetaDataBackup.
Can anybody confirm me that it's ParameterFactMetaData.xml the new file to modify?

In case I'm right, I'm not sure about what to modify.
May I add my new flight mode's name with the value assigned in @ Values (in Parameters.cpp). 
[image]

meee1, 2016-08-12:
ParameterMetaData.xml is automaticlly created during an update, until that point it uses the backup version.

eduherminio, 2016-08-15:
Thanks @meee1, in ParameterMetaDataBackup.xml it does seem something easy to do.
If anyone is experienced in doing this, please have a look at the image attached and confirm that the new flight mode it's only needed to be added as shown, providing the value matches with the one used in your custom define.h.
It does appear in Mission Planner after doing it this way, but just to make sure there's nothing else to add.
[image]

closed 2016-08-16 by meee1

FRLara, 2016-11-02:
Hello!
I am trying to do exactly the same thing.
I created a custom flightmode in arduplane, and I want Mission Planner to be able to correctly select and display it.
I tried adding my custom mode to both the ParameterMetaData.xml and the ParameterMetaDataBackup.xml files, but no results so far. The ParameterMetaDataBackup.xml file (strangely) has duplicated entries for the FLTMODE params, and I modified all of them.

Still, when I connect the pixhawk with custom firmware, my flight mode does not appear in any dropdown menus (Flight Data>Actions, Initial Setup>Mandatory Hardware>Flight Modes, Config/Tuning>Flight Modes), nor in the Full Parameter List params descriptions. When I choose the custom value in the Full Parameter List (for example, setting INITIAL_MODE to the correct value and rebooting), the HUD shows the mode as "Unknown".

What am I doing wrong?
I'm only changing the .xml files in the Mission Planner installed folder, is there somewhere else I should look? Do I need to rebuild the whole code to have the desired effect?

meee1, 2016-11-03:
what you are doing is correct.
please check

my documents > mission planner > ParameterMetaData.xml

FRLara, 2016-11-03:
> please check
> my documents > mission planner > ParameterMetaData.xml

Changing that file did the trick!
Thank you, Michael.

flybabo, 2018-10-12:
I added a new entry to the SERIAL protocol in the ParameterMetaData.xml but the Mission Planner 1.3.58 doesn't seem to care about the xml file.  Is there any way to add a new entry to the parameter as I need it to test the new code supporting the Frsky FPort protocol.

meee1, 2018-10-12:
@flybabo there will be most likerly 3 or 4 copies of the entry, based on the vehicle type. so make sure you modified the correct one.

flybabo, 2018-10-13:
My bad. It works after updating every serial protocol entries.  Thanks!

theonly-unicorn, 2020-07-13:
> 
> 
> what you are doing is correct.
> please check
> 
> my documents > mission planner > ParameterMetaData.xml

my Mission Planner version is Mission Planner 1.3.70,there is no  ParameterMetaData.xml in the path: my documents /mission planner /, how can i display my new flight mode on mission planner ?

ahmadshobake, 2022-01-05:
ParameterMetaData.xml  not working any more , I added the new flight mode in MP source code Common.cs as below :
            else if (firmware == Firmwares.ArduRover)
            {
                //var flightModes = Utilities.ParameterMetaDataRepository.GetParameterOptionsInt("MODE1",firmware.ToString());
                //return flightModes;

                var temp = new List<KeyValuePair<int, string>>();
                temp.Add(new KeyValuePair<int, string>(0, "MANUAL"));
                temp.Add(new KeyValuePair<int, string>(4, "HOLD"));
                temp.Add(new KeyValuePair<int, string>(10, "AUTO"));
                temp.Add(new KeyValuePair<int, string>(15, "GUIDED"));
                temp.Add(new KeyValuePair<int, string>(17, "New Flight Mode"));

                return temp;

            }