ArduPilot/MissionPlanner #1330: Pixhawk and uBlox M8N GPS

Phil1988, 2016-08-13:
Hey guys!

Well I have an M8N GPS that didnt worked out of the box with my pixhawk.
I got no GPS signal at all.
So I used a USB<=>Serail bridge and connected it directly to the computer.
I used u-center and tested it.
With NMEA I get rock solid 3D fixes at all baudrates (tested 38400-115200).
See here:
https://www.youtube.com/watch?v=VypfPqO1P1k

Now that I know that the module itself is working correctly, I tested the ubx protocol.
Sadly this is not stable at any baudrates and measurement periods.
See here:
https://www.youtube.com/watch?v=y2WIN1sDLnE

I then used Mission Planner 1.3.39 and the GPS module connected to pixhawk and get an 3D fix.
But its loses the fix at least one time in a minute.
See here at sec 18:
https://www.youtube.com/watch?v=6GfosbBjnAg

I then took a config that I found here: http://ardupilot.org/copter/docs/common-ublox-gps.html
As soon as I upload the config file via u-center I dont see any signals in u-center. 
I read that this is normal.
The bad thing is, that in Mission Planner it also loses the GPS signal.

I took a look at the "Full Parameter List" in Mission Planner and saw, that I can change the GPS protocol to NMEA. As this gave me stable and good signals in u-center, I configured the M8N again in u-center to the out protocol NMEA with a baudrate of 38400 at 5Hz measurements, Airborne 4g and EGNOS.
This gives me good signals in u-center, but no signals connected to the Pixhawk in Mission Planner.
[image]

So here are my questions:
1. As NMEA protocol is working stable, am I right to say that my M8N is working ok or is it broken?
2. Is there a way to let Mission Planner work with the NMEA protocol?
3. How can I get stable GPS/Glonass signals on my pixhawk with the ubx protocol?

I really did a lot of reading on many forums, guides and did a lot of try and error, but I cant get it working on my Pixhawk.
I hope you can help me.

Thanks a lot!

meee1, 2016-08-15:
sounds like you have a non-gen gps. and nmea does not provide enough information to the autopilot about velocity's.

Phil1988, 2016-08-15:
Thank you @meee1 for your reply.
Sadly I did not understand 2/2 sentences :D 

What is a "non-gen gps"?
Do you mean "generic"?
What can I do about it?

Did you saw my videos?
The video where I tested the NMEA protocol shows solid data from the GPS.
Everything was tested inside and on a window board.

What can I do to solve this problem or what is the problem here?
Thank you a lot and sorry that I'm not professional enough to understand your statements ;)

meee1, 2016-08-16:
a m8n should work in ublox mode with ardupilot without any issues, ardupilot will config the gps on boot. the fact that this is failing indicates a possible hardware issue. your ucenter video in ubx mode, does not look out of the ordinary, as the per satellite data is no required for use with the autopilot.

try turning on all ubx message in ucenter, and verify there.

ie open message view> right click nmea, disable all messages, right cluck ublox, enable all messages.

Phil1988, 2016-08-16:
I connected it directly to the PC with the UART-Serial bridge and did record a cold start:
https://www.youtube.com/watch?v=WlHtFMeHJQU

Is this helpful for you?
Because I cannot verify it (not enough knowlegde)...

You mentioned that the module might be broken.
Did you saw the video I posted above for the NMEA protocol?
Why can I get solid data for this then if it is really broken?

If I can do anything more to help you helping me - let me know :)

meee1, 2016-08-16:
Looking at the snr off the signals you are getting shows that you need a clearer view of the sky. Most of the snr's are bellow 30. Which is not good enough for a reliable gps Fix. The video generally looks fine. The data flashing it to do with the solution rate more than anything else

Phil1988, 2016-08-16:
Yeah... but its a weird thing.
With the NMEA protocol I get solid data (of course weak because inside, but I get a constant data input) with a baud of 38400 at 200ms sending....

With UBX I can only reach constant data at 115200 baudrate and 400ms or slower..
Anything else give me that data flashing (for example 38400 baud with 1000ms) - and I think this causes the "no gps" fault in Mission Planner.
I can see there is an option in Mission Planner to use the NMEA protocol (pic above) but it has NO function at all.

So UBX does only work with a higher baudrate and slower refreshing rate then Pixhawk set it up -  so it doesnt work - and NMEA is not working at all.
I know the signal is not great but good enough for a 3D fix. I also get 3D dgps (whatever this means) with not the best accuracy but good enough to not lose the signal completely.

So how can I get this working (stable) with Pixhawk/Mission Planner?

Phil1988, 2016-09-02:
So noone has additional hints or ideas how I can make it woking?
I found it really irritating, that there is a UBX option in Mission Planner that doesnt work.