ArduPilot/MissionPlanner #1332: Full Parameter Tree can't write more than one parameter at a time

rmackay9, 2016-08-17:
It seems that if I update 3 parameters on the full parameter tree page and then push the Write Params button they are not saved.  Also even if I tab away and back to the page, or push refresh, the green highlighting remains around the changed parameters.
If I shut down the mission planner, restart and write then individually I can save them.
[image]

I'm using the latest beta Mission Planner and Copter-3.3.3.

meee1, 2016-08-18:
fixed.

closed 2016-08-18 by meee1

rmackay9, 2016-08-22:
Thanks.  I've tested and it works.

One issue I've noticed is that pressing the "Refresh Params" button doesn't clear the green highlighting.  Maybe I'll create a new issue for that.