ArduPilot/MissionPlanner #1345: MP - Unable to do intial setup - Accel calibration unsucessful

lvale, 2016-08-26:
MP 1.3.39.2 build 1.1.6081.19259

Latest Copter 3.4dev PX4-V4 from 8/24 

Tested on 2 different PC's running Win7 and Win10

Connection via USB cable (used different cables)

Unrelated problem - was unable to flash Custom file because MP never managed to find the PixRacer to flash it. So used other method to flash it.

So I returned to MP to try to do the Initial setup (didn't use the wizard)

PixRacer connected ok and detected on COM5.

While trying to do Accel calibration the system takes a long long time to change from level to left to right, etc and fails with a message Accel failed.

This blocks any other calibrations required.

Attached tlog from the last failed session

[2016-08-27 00-22-30.tlog.zip](https://github.com/ArduPilot/MissionPlanner/files/440327/2016-08-27.00-22-30.tlog.zip)

meee1, 2016-09-05:
where you connected via usb or radio?

meee1, 2016-09-05:
according to the log, and what mission planner saw

[image]

this to me looks like an ArduPilot issue, but would like you to confirm the usb/radio first.

meee1, 2016-09-05:
what the SS shows is that the command_ack are you pressing the button. and at  23:15 you press the button, but the autopilot doesn't request the next state. now if the packet was lost then when you press the button at 24:12 the autopilot should have again displayed a message, but it didn't. also the autopilot should be sending the message on a timer (it used to), as there is no guarantee there wasn't packet loss on the receive side.

in conclusion its a ArduPilot bug, as for the cause, not sure

closed 2017-01-02 by lvale