ArduPilot/MissionPlanner #1346: MP - Unable to do setup Polygon fence

lvale, 2016-08-27:
Latest release of MP beta with latest master of Copter 3.4

Unable to create Polygon fence

[image]

magicrub, 2016-08-27:
"Latest master" or "current beta" doesn't help when this is seen 2
weeks/months/years from now. What version number?

lvale, 2016-08-27:
Top bar of MP
Copter 3.4dev from 24-8-2016

meee1, 2016-08-27:
I just tested and everything looks fine.
did you connect via a non standard method?
            if (MainV2.comPort.MAV.param["FENCE_ACTION"] == null)
            {
                CustomMessageBox.Show("Not Supported");
                return;
            }

also could be the FENCE_ENABLE issue.

lvale, 2016-08-28:
Connection was via USB on this screenshot, because I had tried previously via telem radio also unsucessfully.

closed 2016-08-29 by lvale