ArduPilot/MissionPlanner #1347: Feature Request - Add Solo to firmware upload

jmachuca77, 2016-08-28:
As part of a new initiative to support Solo on master, we need to support a way of uploading custom firmware to Solo. The process is fairly simple, the firmware file should be uploaded into /firmware on Solo, and then the command loadPixhwak.py should be issued in Solo. 
Solo_Cli handles this on the file flash.py function flash_px4. They also erase the contents of the folder /firmware/loaded.

meee1, 2016-09-18:
@jmachuca77 would the plan to be to provide a custom file for upload?

proficnc, 2016-09-18:
to start with, just normal or custom pixhawk firmware, but later, the ability to also interact with the co-computer.

lroberts99, 2017-02-10:
Is there any update on this? I am trying to upload custom firmware to the Solo with no success.

rmackay9, 2017-02-11:
@lroberts99, you've seen this wiki page? http://ardupilot.org/dev/docs/solo.html

I haven't tried them recently but they worked at the time that they were written (>1year ago).

lroberts99, 2017-02-11:
@rmackay9, yes, I have followed the instructions on that page. I get an error when trying to make the .px4 file from the ardupilot-solo code. The make process seems to require a config.mk file and it gives me this error: `../mk/px4_targets.mk:30: *** ERROR: NUTTX_SRC not set correctly - no configs directory found. Stop.` Maybe with @meee1's recent commit the upload will work. I will update with any progress next week.

proficnc, 2017-02-11:
Are you having issues compiling? Or uploading?

And if you are compiling the solo version of Ardupilot, then you need to raise issues there.  That is a fork, and is way out of date.

jmachuca77, 2018-07-20:
This has already been implemented in Mission Planner

closed 2018-07-20 by jmachuca77