ArduPilot/MissionPlanner #1348: FlightGear Simulation and HIL support removed from current build of Mission Planner

kngako, 2016-08-30:
"The issue for us with SITL vs HIL is where the code runs: in SITL, the code runs on the PC, and not on the APM. In HIL the code runs on the APM."

Testing APM with Flightgear can no longer be implemented since the code was removed in the commit 7044c1a6509a84698bda9a833e4fd077364ce2e5. 

Uploading HIL Simulation also isn't seen anymore on the Install Firmware Tab. 

How will these two functionalities be supported in future?

meee1, 2016-09-05:
they are being replaced by sitl. the current sim of choice is also x-planes.

Support for flightgear might return, support for HIL will not return.

kngako, 2016-09-05:
Hmm... thanks for the heads up.

kngako, 2016-09-05:
Can I ask what are the reasons for taking it out?

meee1, 2016-09-05:
because support from the firmware is being removed.
this all has to do with the amount of data that needs to be sent between the pc and the autopilot to make it work. the connection is not fast enough.

closed 2016-09-12 by kngako