ArduPilot/MissionPlanner #1350: Inspire 1 Pro camera listing

MarkDFink, 2016-08-31:
I just upgraded to version 1.3.39 and see in the default cameras listed five entries for DJI Inspire 1 Pro using various lenses. Does this mean that missions created in Mission Planner can be used on the DJI Inspire 1 Pro platform? I also see that there are two options for saving waypoint files: the usual *.txt and a new Mission JSON *.mission format. Are these updates related to each other? Or is this implemented so that you can use the Inspire 1 Pro X5 camera on an existing hexacopter?

meee1, 2016-09-05:
what camera are you referring too?

MarkDFink, 2016-09-05:
Hi Michael,

 

Here is a screen shot with the Inspire cameras from the Survey screen. I’ve also included the camera xml file.

 

Thanks,

Mark

 

Mark Fink

 <http://www.Virtual-Travels.com> www.Virtual-Travels.com

1-845-380-0068

Explore A Virtual World

meee1, 2016-09-18:
no screenshot.

MarkDFink, 2016-10-27:
Sorry for the delay. We were away on vacation. I realize I didn't attach the screenshots to the thread, but attached them to the email. Let me know if you see them now.

Thanks,

Mark
[image]
[image]

MarkDFink, 2016-10-27:
Oops - hit the Close button by accident...

meee1, 2016-10-28:
where did you get your copy of mission planner from? the default version does not have those cameras

MarkDFink, 2016-10-28:
It was from an online automatic update. Was there a version online that shouldn't have been? Does this mean you have a version in development that supports the DJI platform?

MarkDFink, 2016-12-01:
Any news on this? I would be VERY interested in continuing to use Mission Planner with my DJI. I've been testing several other software packages, and yours is the best. No contest.