ArduPilot/MissionPlanner #1353: "control-T" with multi-vehicle connections and background param fetching

nicksargeant, 2016-09-09:
When connecting to a UAV over a poor link (i.e. excessive packet loss), it can take a very long time to pull down the parameters. It is possible to connect to an aircraft without having to fetch parameters by pressing control-T instead of clicking Connect.  
A couple of minor requests:

1.) Instead of clicking control-T, it would be great if there were a checkbox next to connect with the option "do not fetch params" (or equivalent)
[image]

We have been doing a lot of multi-vehicle flying recently (GCS1<-915(primary)->Plane<-433(secondary)->Multirotor), see: https://www.facebook.com/PerthUAV/videos/1466195816728838/.
Because this topology requires Mavlink data from both aircraft to be routed through the primary link (not helped by the fact that typically the second link isn't very strong), when selecting to control the second aircraft MP first fetches all of the parameters from the second aircraft (which takes a long time and stops the user being able to do anything else with the GCS) .  SRX_* parameters (except SRX_PARAMS) have been adjusted to reduce amount of non-critical transmitted along link. 

2.) It would be great if it were possible for connect to the second aircraft without having to pull its parameters [following on from the above checkbox request, if implemented then do not pull params from any aircraft connected to the GCS].
On a side not, a couple of things in MP have changed since the multi-vehicle documentation on the Wiki was written so I am happy to put my hand up to help revise the documentation.

3.) Finally, possibly a more longer term request; would it be possible for the parameter fetching to be conducted in the background so that a user can still monitor and control the UAV when connecting?

Thanks as always!

meee1, 2016-09-11:
ive added a option to bypass the param get if you hold control while changing the vechicle, this is the immediate fix. I will work on a longer term fix

meee1, 2016-09-11:
with your point 1, make sure you adjust the missionplanner stream rates as well, in the planner settings, as these technically override the srx_ values

meee1, 2016-09-11:
also on a side note, you can click cancel on the param fetch when you change vehicles, and it shouldn't matter. unless you want to goto the config/tuning pages, as they enable/disable based on what parameters are available.

meee1, 2016-09-17:
1.3.40 add background param polling, for the currently selected vehicle. it will still try and get the entire param list on vehicle change, unless you hold control while changing. Also clicking cancel is also an option.

nicksargeant, 2016-09-18:
Brilliant! I will test this evening. Once I understand this functionality, I will update the multi-vehicle wiki docs. Thank you very much