ArduPilot/MissionPlanner #1354: Terrain following on specific waypoints

ThomasDmr, 2016-09-13:
Hi, 
First of all, thanks for the amazing work ! 
I've been playing around with the arducopter 3.4 beta and the latest version of mission planner. There is this great new feature of terrain following, but it only applies to the entire mission.
Is there a way to have the terrain following activated only on specific waypoints (those close to ground) and not others (those high in the sky) ?

What I need to do is plan a mission where a drone flies some distance above 50 m where just the baro is enough,but then I would likea waypoint at 2 m above ground and would like the drone to use the sonar to be sure it doesn't crash.

Is there any way to implement that through mission planner ? 

Thanks a lot for the help and good luck.

meee1, 2016-09-17:
this is not currently possible with mission planner, and I personally don't like this option, as it can create a very bad situation.

ThomasDmr, 2016-09-18:
Hi, 
Thanks for your answer. Could you please explain why it might create a very bad situation?

meee1, 2016-09-18:
because the use of relative, terrain and absolute in one mission leads to nothing but confusion. While it is possible to do, it can create a bad for user experience. Also I think what you are trying to do is already possible. you would need to ask that specific question on the copter forum

ThomasDmr, 2016-09-19:
I'll ask about it on the arducopter forum.
Thanks for your help

rmackay9, 2016-09-19:
So currently with Copter-3.4 it's not possible to specify terrain altitudes (in the mission) that are above the range of the lidar (assuming a lidar is being used instead of the google earth data). Tridge and I have discussed that, if the user specifies a terrain altitude that is definitely above the lidar's range we could switch to using google earth data but we need to test that because there can be significant differences between the two (I've seen google earth data that is >30m off of reality).

The limitation of mixing terrain-alt with relative or absolute altitudes is purely a limitation in the GCSs.  It may confuse people to allow both but as far as I know, the underlying ardupilot Copter code supports this.