ArduPilot/MissionPlanner #1361: Display distance to Geofence (as a user item)

nicksargeant, 2016-09-18:
Is is possible to dynamically calculate and display the distance between the aircraft's position and the closest point of a geofence? I appreciate that this may be a complicated thing to ask for so understand if it is not high on the priority list (or perhaps this would be best done autopilot side?). 
Thanks.

nicksargeant, 2016-09-26:
Thanks @meee1, I just saw that this was included in the latest release. Will test it this evening.

meee1, 2016-09-26:
@nicksargeant I believe there are no bugs, but use with caution.

meee1, 2016-09-26:
@nicksargeant also, currently there is no indication if its inside or outside the fence, just a number indicating distance to the fence

meee1, 2016-09-30:
todo: fix possible issue if outside the fence on a corner. will currently fail