ArduPilot/MissionPlanner #1362: Prefetch maps problems

lvale, 2016-09-18:
Map pre fetch should/could be a bit smarter.

In general it should be restrained to cache a user determined min and max zoom level. Also when canceled it should abort the entire process, not only the caching of the current zoom level.

When caching a WP path is worst because it goes "wild" and tries to cache ALL zoom levels for ALL the WP points, and completely ignores the user request to cancel the process....

Example here

https://www.youtube.com/watch?v=0GblrvrkP7s&feature=youtu.be

jagrandle, 2016-09-18:
Agreed, being able to preset a maximum zoom would mean that it could be set and left to run without constant supervision. However, the "Cancel" button is the only current way I am aware of stopping pre fetch once you have reached the desired zoom level. You would need to implement that max/min function before changing the effect of the cancel button otherwise there would be no way to stop it at the desired zoom level without wiping the maps you had just downloaded, or letting it go to the end which is usually not desirable (I have never done it !).

meee1, 2016-09-19:
the max internal zoom is 21, it will stop by itself at this level.

ive fixed the path cache by going around the location then zooming in. This feature was added in a rush some time ago, and I never revisited till now.

lvale, 2016-09-19:
@meee1 I would suggest to implement something like @billbonney or Don Gagne did on APM Planner and QGC. Define Area, Initial zoom level, Final Zoom level.

meee1, 2016-09-30:
i believe this is all done now.

closed 2016-09-30 by meee1

lvale, 2016-10-03:
Well, now it completely ignores the max zoom that we set it to.

https://youtu.be/eGbt6rQoWGI