ArduPilot/MissionPlanner #1363: System.MissingMethodException: Method not found: 'Void MissionPlanner.Controls.HUD.set_gpsfix2(Single)

clovett, 2016-09-22:
Fresh install of Mission Planner on clean Windows 10 machine and I get this error:

"A Major error has occured : System.MissingMethodException: Method not found: 'Void MissionPlanner.Controls.HUD.set_gpsfix2(Single)'.
at MissionPlanner.GCSViews.FlightData.InitializeComponent()
   at MissionPlanner.GCSViews.FlightData..ctor() in C:\Users\michael\Source\Repos\MissionPlanner\GCSViews\FlightData.cs:line
175
   at MissionPlanner.MainV2..ctor() in C:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:line
652"

meee1, 2016-09-22:
@clovett  msi or zip? I assume msi.

meee1, 2016-09-22:
the gac isn't coming into play here either?

clovett, 2016-09-24:
uninstall and reinstall (using this msi link) into a clean directory now works, thanks (it is possible I installed newer bits into a directory that already existed and that may have caused problems).

closed 2016-09-24 by clovett