ArduPilot/MissionPlanner #1364: AltitudeAngelWings build error.

clovett, 2016-09-22:
1>C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets(4106,5): error MSB3030: Could not copy the file "D:\git\MissionPlanner\ExtLibs\AltitudeAngelWings\Keys.config" because it was not found.

magicrub, 2016-09-22:
does that file exist?

meee1, 2016-09-22:
its not included on purpose, check .gitignore. however I think I may need to add some placeholder, as this I now the seconds notice I have received about this

clovett, 2016-09-24:
They way I do private machine signing is to create an environment variable on that machine which is picked up by VS like this:
<AssemblyOriginatorKeyFile>$(MyKeyFile)</AssemblyOriginatorKeyFile>

This way if that variable is not defined then the project builds just fine, just won't be signed.  For app.config you could potentially create the included keys.config from MSBuild when some environment variable is set.  By the way I couldn't see where that MapKey is actually used in the code, and there's another mapkey in Mapbox.cs.

meee1, 2016-09-30:
this should be resolved now. remove ref from project file

closed 2016-09-30 by meee1