ArduPilot/MissionPlanner #1365: VS F5 run after upgrade fails

clovett, 2016-09-22:
Moving file from 
D:\git\MissionPlanner\bin\Debug\config.xml
to
d:\Users\Chris\Documents\Mission Planner\config.xml
fails
Additional information: Cannot create a file when that file already exists.

Line 233 of Settings.cs.

Perhaps the answer is to do a copy then delete the old file to be sure we don't reenter this code path.
                File.Copy(path,newpath, true);
                File.Delete(path);

magicrub, 2016-09-22:
does a clean build work?

meee1, 2016-09-22:
@clovett can you report back as to what state the files are in?
considering the first run is a move, I'm not sure how the file can still exist in the original location. if you can highlight why that may be that would be great.

meee1, 2016-09-22:
@clovett never mind, "multiple installs" is the concept I was not thinking of.

closed 2016-09-24 by clovett