ArduPilot/MissionPlanner #1366: Aviation Maps

Jman841, 2016-09-23:
Would it be possible to overlay aviation VFR Maps such as Skyvector into mission planner to verify the type of airspace you are flying into?

The current Part 107 does not have any restrictions on a 5 mile radius from airports, only if you are flying in other than Class G airspace do you have to get in touch with the local airports to notify. It would be great if mission planner could overlay these maps of where your mission is to make sure you entire mission is not entering into any other type of airspace that would require notification.

magicrub, 2016-09-23:
+1
would also be nice to bring up a warning if within certain airspaces.

iskess, 2016-09-23:
+1 this would be great. With the addition of ADSB, this shows us likely flight patterns of other aircraft. We will see airways, military training routes, the floor of airspace rings. Also hangglider and glider areas. Now that FAR107 is here, airman have knowledge of the sectional chart.

meee1, 2016-09-24:
firstly, I'm in Australia, 107 means almost nothing to me.
secondly, is there a open source of this data?
thirdly, what format is the said data in?

got no issues adding support, but definitely need more info.

meee1, 2016-09-24:
Altitudeangel might provide some of this info as well, and support is in beta MP for this.
control-f > altitude angel (need to sign up)

magicrub, 2016-09-24:
Good point, Michael. Altitude Angel is a good solution to this. Let's see
how that goes but meanwhile let's keep this issue open.

rupertbenbrook, 2023-07-20:
We don't have any plans to include VFR maps as they're of little help to Drone flights and contain much information that not relevant for drones that isn't in our map data in other ways.