ArduPilot/MissionPlanner #1368: 1.3.41 font size and viewing is all messed up

Jman841, 2016-09-24:
Not sure what happened, but the font size and boxes and everything in the latest released is all off and I can't see many things. Photos below...

https://www.dropbox.com/s/kxbl3g55verecgd/Mission%20Planner%20Error%201.jpg?dl=0

https://www.dropbox.com/s/jvttezpnj36lqbj/Mission%20Planner%20Error%202.jpg?dl=0

meee1, 2016-09-24:
im going to guess that you change the font scale size on your pc.
[image]
please change the text size back to 100%

Jman841, 2016-09-24:
Very Strange. I've never had that issue with mission planner before. I just changed it to 100% then back to 125 where I have it and now it is fixed. 

Any idea why this update had that issue? I'll close this as it's probably an isolated incident

Jman841, 2016-09-24:
Something is

Jman841, 2016-09-24:
Something is definitely off with the latest version. I Closed mission planner, reopened it, and it has the same issue. I've never had this issue with previous versions.

meee1, 2016-09-24:
did your pc do updates? as MP hasn't change anything that would cause a font change

Jman841, 2016-09-24:
Very strange, There was the big windows 10 anniversary update about a week ago, however, I used 1.3.40 and 1.3.39.1 after that without issues. Sometimes it opens perfect and everything looks good, then randomly it resizes everything. Not exactly sure what's going on, but, something is not happy with windows and the latest version of mission planner.

gwebb1, 2016-09-25:
I believe I'm running into the same or similar odd issue as described above, any suggestions for fixing it besides turning off DPI scaling? Unfortunately my laptop screen is 13" with a 3200x1800 resolution which means text is extremely small without 200% DPI scaling.

Version 1.3.41 starts up okay but the DPI scaling looks like it breaks shortly after connecting to a Pixhawk. Right when the connection dialog box says "Getting Params", Immediately after the "Mavlink Connecting" and "Done" messages are shown, the main mission planner window re-flows with most but not all text shrinking and partially moving off screen.

When I installed 1.3.40 again, the issue went away but came back when I upgraded to 1.3.41 again.

andrewmoni, 2016-09-25:
I have exactly the same problem on my surface 3 running latest windows 10.  Also 1.3.40 was fine - problem has occurred only in 1.3.41 and renders mission planner useless.  Starts fine, and resizes when connecting to pixhawk.

andrewmoni, 2016-09-25:
As an add-on to previous comment, I also run with zoom set at 155% as 100% renders the writing far too small.  How can I go back to 1.3.40???????

meee1, 2016-09-30:
latest beta should resolve this.

andrewmoni, 2016-09-30:
Where do I find beta release of MP?  I know I can get beta release of copter from within MP - but where do I find beta of MP - or do you mean 'latest version' instead of 1.3.41 - which then begs the issue of when can 1.3.42 (with the correction to 1.3.41) be expected, thus possibly remove the need to try a beta of MP!!??

meee1, 2016-09-30:
open MP > help > update beta

andrewmoni, 2016-09-30:
Doh - silly me - never noticed that beta load button before!!  Yes, 1.3.41.3 build 1.1.6116.13955 seems to work fine - many thanks for the quick response.

Jman841, 2016-10-02:
Thank you, Seems like it's fixed. I'll close this one down.

closed 2016-10-02 by Jman841