ArduPilot/MissionPlanner #1370: Log Analyser crashes runner.exe

lvale, 2016-09-28:
When trying to execute Auto Analysis, from the DataFlash tab, Log Analyser /runner.exe crashes

Latest beta, as in the screenshot

Logs are from Copter 3.4RC5 and trying to use the log file (not the bin)

[image]

meee1, 2016-09-30:
does it happen on all 3.4-rc5 logs?

lvale, 2016-09-30:
Hi, sorry but forgot to mention that the logs were from 3.4RC5. I try to analyse a converted log, not a bin, if that helps.

estebanavv, 2016-10-02:
This happens to me too using Mission Planner 1.3.41 and the log from ArduPlane V3.5.3 (b8461671) .bin log

Fossdal, 2016-10-05:
same at this bug #1315

magicrub, 2016-10-05:
Luis, please close this issue on favour of the other original issue

lvale, 2016-10-06:
Closed because similar to https://github.com/ArduPilot/MissionPlanner/issues/1315

closed 2016-10-06 by lvale