ArduPilot/MissionPlanner #1374: Altitude Angel visibility

lvale, 2016-10-02:
If enabling Altitude Angel the overlays are too dark completely hiding map

Also, didn't find a way to disable it

[image]
[image]

meee1, 2016-10-02:
@lvale can I get a approx. coords for that location?

meee1, 2016-10-02:
as for disable, latest beta > planner settings > Altitude angel button

OXINARF, 2016-10-02:
@meee1 That's around Lisbon, Portugal.

aercamti, 2018-12-10:
I have been battling with the same issue. 
We recently had filed a NOTAM, which made the flight area appear blue. Yet if you have to operate within that area yourself, you can't see any satellite imagery anymore, unless completely turning off Altitude Angel (but this is not desirable when operating in complex airspace).

@meee1 The more elegant and useful way would be to have an opacity slider for each overlay, accessible right from the main interface, to be able to dynamically adjust the transparency of a layer. 
A good example: https://map.geo.admin.ch/?topic=ech&lang=en&bgLayer=ch.swisstopo.pixelkarte-farbe&layers=ch.bazl.luftfahrtkarten-icao&catalogNodes=687,702&layers_opacity=0.55 
(Click the "gear" icon next to the ICAO chart to adjust)
_Now_ this may mainly be helpful for Altitude Angel uses - yet in the future this would also be good for other custom layers needed for flight planning.

Thanks for considering to add this.

P.S.: Altitude Angel is an official cooperation partner for NATS UTM. Use of such air navigation services is likely to only increase, as regulations will become more stringent, at least in Europe, as early as 2019.

rupertbenbrook, 2023-07-20:
@meee1 & @lvale. Fixed in recent builds as we've added an opacity setting for the map layers. Right click the map and choose "Altitude Angel Settings" then the Map tab.