ArduPilot/MissionPlanner #1385: altitude angel sign up issue

tridge, 2016-10-07:
saw the new altitude angle feature today!

got this after signing up:

Runtime Error 
Description: An application error occurred on the server. The current custom error settings for this application prevent the details of the application error from being viewed remotely (for security reasons). It could, however, be viewed by browsers running on the local server machine. 

Details: To enable the details of this specific error message to be viewable on remote machines, please create a <customErrors> tag within a "web.config" configuration file located in the root directory of the current web application. This <customErrors> tag should then have its "mode" attribute set to "Off".

<configuration>
    <system.web>
        <customErrors mode="Off"/>
    </system.web>
</configuration> 

Notes: The current error page you are seeing can be replaced by a custom error page by modifying the "defaultRedirect" attribute of the application's <customErrors> configuration tag to point to a custom error page URL.

<configuration>
    <system.web>
        <customErrors mode="RemoteOnly" defaultRedirect="mycustompage.htm"/>
    </system.web>
</configuration>

meee1, 2016-10-19:
ive sent this to AA, as this is a server side issue on their side.

rupertbenbrook, 2023-07-20:
@meee1 and @tridge, I believe this is no longer a issue. Can we close and reopen if it's still a problem?