ArduPilot/MissionPlanner #1388: Joystick config does not save or load properly

nanthony21, 2016-10-15:
I can see that the joystickbuttons.xml and joystickaxis.xml files save properly in User/Documents/Mission Planner but when the configuration window is opened it is always completely unset, just like the first time it was opened. 

It also seems that the .xml files are only properly saved to if the "autodetect" feature isn't used.

meee1, 2016-10-19:
are you connected to anything when you config the joystick? MP stores separate joystick configs for plane and copter.

gediminasgu, 2016-10-20:
I can confirm that Joystick panel doesn't show saved values, even seems like mapping is stored correctly. See the video: http://d.pr/i/uqTo
MP version: 1.3.41 build 1.1.6110.31287

lakshaydang, 2018-01-31:
I am also not able to save the values that i configured for my joystick. Help me !