ArduPilot/MissionPlanner #1389: Overlaping of quadcopter and plane

pedroalmoutinho, 2016-10-15:
Hello,

I am seeing an issue that I can not solve. I already googled but I stil have no solution for this weird situation.

In the initial screen appears a quadcopter and a plane, but the versian is for Plane.

Can anyone tell me what am I doing wrong.
[image]

magicrub, 2016-10-16:
Have you tried latest mission planner and latest arduplane?

pedroalmoutinho, 2016-10-17:
Finally I solved the problem. The problem was not with Mission Planner. I also tried with MavProxy and I also noticed the problem.

The problem was that I was using a 3dr radio for telemetry, but now I changed to a laird module (2.4ghz), like XBee. I forgot to disconnect tx from telemetry. With Xbee radios just connects rx, ground and +5v.

My fault, sorry!!

closed 2016-10-17 by pedroalmoutinho