ArduPilot/MissionPlanner #1391: Mission Planner v1.3.41 script import numpy error

spmish, 2016-10-18:
I'm getting the following error when trying to use the numpy library in a simple practice script. 
"Error running script cannot import multiarray from numpy.core"

Here are the basics of my setup:
   OS: Windows 7 Pro (64-bit) 
   Mission Planner v 1.3.41
   Python 2.7 (32-bit) install
   numpy (32-bit) install for windows Python 2.7 installed with pip, no errors.

My simple practice script is below:

import sys
sys.path.append(r"c:\Python27\Lib\site-packages")
sys.path.append(r"c:\Python27\Lib")
import numpy as np
a = np.array([1,2])

Any suggestions?

meee1, 2016-10-19:
mission planner will most likely require the 64 bit version of python and its libs

spmish, 2016-10-19:
Thank you, so much for the feedback. I went ahead and removed the 32 bit versions of Python2.7 and all libraries. Then installed the latest 64-bit versions. Unfortunately, I'm still getting the same error message when I run the above script with the numpy library. I've attached a screenshot that has my IDLE editor overlaid with the mission planner error just to show the revisons of each. Other scripts that do not call numpy seem to be running fine. 

Here is a screenshot of the error if it helps. 

[image]

Thanks again for your continued help!

dongsin-kim, 2020-05-19:
Hi @spmish,

I understand that you were trying to following the reference code (for [Using CPython Standard Libraries](https://ardupilot.org/planner/docs/using-python-scripts-in-mission-planner.html))

1. Could you manage to handle this error?

2. Wasn't the letter *r* between the letter *(* and *"* typo?
sys.path.append(r"c:\Python27\Lib\site-packages")

I look forward to hearing from you.

Dongsin