ArduPilot/MissionPlanner #1393: Command coordinates precision vs. ArduPilot log

ThomDietrich, 2016-10-21:
I'm not sure where to file this issue.

MissionPlanner stores waypoints/commands with coordinates in the following way (example):
```
7	0	3	16	0.000000	0.000000	0.000000	0.000000	51.077282	10.988579	20.000000	1
```
ArduPilot later creates logfiles with the following content:
```
CMD, 42187994, 9, 7, 16, 0, 0, 0, 0, 51.07728, 10.98858, 20
```

As you can see, the **precision differs** by one decimal place. One could argue if this is a bug or not but it for sure confused me while doing a custom analysis. The additional MP decimal suggests more than ArduPilot can handle. Would you consider removing it?

meee1, 2016-10-25:
> Thanks for the explanation. What's your opinion on that matter?
>
> https://github.com/ArduPilot/MissionPlanner/issues/1393#issuecomment-255429891

Please check this link https://en.m.wikipedia.org/wiki/Decimal_degrees

magicrub, 2016-10-25:
Mission Planner uses Double Precision. MAVLink usually uses float (athough
integer is available and better). Ardupilot logs are float. This is a side
effect of a float to integer transition on Ardupilot for improved RTK
accuracy.

magicrub, 2016-10-25:
The sooner we can convert to double natively on Ardupilot, the better.
However, that eats up lots of ram and slows down computation as well as
increases demand on external flash memory and its bus. This would be a more
natural progression to a Linux only benefit due to limited external flash
memory on pixhawk.