ArduPilot/MissionPlanner #1398: Mission planner unusable on HiDPI displays

jcooper21, 2016-10-24:
Not sure what happened in these last updates but mission planner is now completely unusable on a Microsoft surface pro 4. All of the text and formatting is completely messed up.

magicrub, 2016-10-24:
What version is "latest version"? There was a problem on v1.3.41 which was fixed via this issue:
https://github.com/ArduPilot/MissionPlanner/issues/1368

jcooper21, 2016-10-24:
It's on the latest stable version.

jcooper21, 2016-10-24:
It only happens when the vehicle connection is established.  When it starts up and is not connected it looks normal.

magicrub, 2016-10-24:
Please specify a version number. "Latest" doesn't mean much to developers who are juggling several versions at once

jcooper21, 2016-10-24:
version 1.3.41 build 1.1.6110.31287

magicrub, 2016-10-25:
Please provide a version number.

bluesign2k, 2016-10-28:
I can confirm that I'm having the same issue. I saw elsewhere that updating to the latest beta (at time of writing this is 1.3.41.3 build 1.1.6145.17836) might fix it but now the program loads up and very briefly shows everything correctly then almost immediately re-scales to the same mess (see pic for example).
[image]

I'm running it on a Surface Pro 4 that has a native resolution of 2736 x 1824 using the recommended scaling of 200%. I have come across a few other programs that behave similarly and have been able to hack a fix (using the instructions here: http://www.danantonielli.com/adobe-app-scaling-on-high-dpi-displays-fix/) but that doesn't seem to work for mission planner for whatever reason.

kikislater, 2016-10-29:
Same for me on (Teclast X98 air ii) 10" tablet which have 2048 x 1533 native resolution.
Resizing window change police size. 
But when I connect to vehicule mission planner is resized and with very very small fonts. When mavlink telemetry finished, I resized the window, fonts are still small, touching flight plan and fonts began better and everything is ok. I could make a video if you want

wjax, 2016-11-11:
Same here.
Please Michael, make it DPI-Aware or at least, disable in the manifest the dpi-aware and force scaling.

wjax, 2016-11-11:
Seems fixed in beta 1.3.41.6

Will report if happens again.

bluesign2k, 2016-11-12:
Following the above comment, I've just asked it to update to the latest beta. It grabbed 1.3.41.7 build 1.1.6160.31408 and I can confirm that it's still very much broken for me.
Rolling back to 1.3.40 fixed things again.

lvale, 2017-12-05:
@meee1 

using a open issue to report that the "trick" to have HighDPI friendly with MP 1.3.50 is not working IF you install MP from the Store.

[image]

meee1, 2017-12-05:
@lvale I will most likerly be pulling it from the store, due to the fact I cant upload firmware with the store version

10.2.4
Your app may contain fully integrated middleware (such as third-party cross-platform engines and third-party analytics services), but must not deliver or install non-integrated third-party owned or branded apps or modules unless they are fully contained in your app package.
Your app may depend on non-integrated software (such as another app or module) to deliver its primary functionality, subject to the following requirements:
You disclose the dependency at the beginning of the description metadata
The dependent software is available in the Store

phelix001, 2022-05-02:
https://imgur.com/a/OQiTYky 
Linux billygates 5.16.19-76051619-generic #202204081339~1649696161~22.04~091f44b SMP PREEMPT Mon Apr 11 17 x86_64 x86_64 x86_64 GNU/Linux

On a yoga c930 with highdpi touch screen, running most recent MPlanner.exe witih mono. 

Any ideas or solutions?

dividebysandwich, 2024-05-22:
It's 2024 and the issue persists across platforms. This is especially problematic on mobile rugged devices that one would use in the field, like a Dell Rugged Extreme 7212. On Linux Mint my screen looks exactly like @bluesign2k 's screenshot.