ArduPilot/MissionPlanner #1404: LOITER_TURNS Fixed wings

JeyPi, 2016-10-26:
This is how I set the mission for LOITER_TURNS.
My plane flies to the loiter position, where the 1 parameter is set to 5 turns. But it does not loiter - It flies through it to the next way point. Is that a bug?

MAV_CMD_NAV_LOITER_TURNS	Loiter around this MISSION for X turns
Mission Param #1	Turns
Mission Param #2	Empty
Mission Param #3	Radius around MISSION, in meters. If positive loiter clockwise, else counter-clockwise
Mission Param #4	Forward moving aircraft this sets exit xtrack location: 0 for center of loiter wp, 1 for exit location. Else, this is desired yaw angle
Mission Param #5	Latitude
Mission Param #6	Longitude
Mission Param #7	Altitude

JeyPi, 2016-10-27:
It looks that the waypoints I set were too close there fore the mission was not flown complete

closed 2016-10-27 by JeyPi

magicrub, 2016-10-27:
also not a Mission Planner GCS bug. If it was a bug, it would be an [ardupilot issue](https://github.com/ArduPilot/ardupilot/issues).