ArduPilot/MissionPlanner #1405: two commands done simultaneously?

JeyPi, 2016-10-27:
I would like my plane to enter loiter during which it would multi-drop cargo.
At the moment I think it is not possible to achieve as the commands are executed one after another.
I am using Misson Planner standard command list. 

 
LOITER_TURNS     /2/              (what to do have the autopilot execute the following drops during the loiter) 

DO_SET_SERVO     /7/   /1200/     (Drop 1)

CONDITION_DELAY  /3/

DO_SET_SERVO     /7/   /1450/     (Drop 2)

CONDITION_DELAY  /3/

DO_SET_SERVO     /7/   /1700/     (Drop 3)

CONDITION_DELAY  /3/

DO_SET_SERVO     /7/   /1950/     (Drop 4)

Written like this the autopilot executes the two turns and than each three seconds does one of the 4 drops.
I would like to have the drops done during the loiter. Not before or after but during the execution of the loiter turn command. Any idea how to achieve that?

magicrub, 2016-10-27:
Hi @JeyPi you've posted a help/HowTo question in the Mission Planner bug tracker and feature request system. I have a possible solution that I'll post here but please close this issue and use discuss.ardupilot.com and link the conversation here so i can find it if you need further help. The secret is all NAV commands block until they complete and all DO commands are instant.

Use NAV_LOITER_TIME instead of delay:

NAV_LOITER_TIME (1 minute) - blocks here and takes 1 minute to complete 
DO_SET_SERVO, 7 , 1200 - drop 1
NAV_LOITER_TIME (3 seconds) - continue loitering for 3 seconds 
DO_SET_SERVO, 7 , 14500 - drop 2
(... repeat ...)

magicrub, 2016-10-27:
moved to http://discuss.ardupilot.org/t/arduplane-executing-two-command-simultaneously/12403

JeyPi, 2016-10-28:
Hi yes it worked thanks a lot. Closing. I will post some pictures of the mission planner screen and the multi release system at the link mentioned above

closed 2016-10-28 by JeyPi