ArduPilot/MissionPlanner #1406: Log Downloader shifted enumeration of logs

ThomDietrich, 2016-10-28:
As can be seen in the screenshot, logged flights are enumerated starting from 1 in the left pane but seem to be enumerated starting at 0 internally as can be seen in the right pane where the error occurs for log 6 / 5.
Suggestion: adapt the left pane to also enumerate starting at 0.

[image]

(This issue is not related to the warning seen in the screenshot)

magicrub, 2016-10-28:
I don't see that in Ardupilot, so the index problem must be contained within Mission Planner.

ThomDietrich, 2016-10-28:
That's what I think too. Hence the issue under MissionPlanner ;)

meee1, 2016-10-31:
ive checked this out, and it is just a cosmetic bug, the correct log was downloading.