ArduPilot/MissionPlanner #1407: Wrong speech dialogue

kikislater, 2016-10-28:
Mission planner is speaking about Australian dollar ! 
https://www.youtube.com/watch?v=Rk9pLw3132s

magicrub, 2016-10-28:
It's silly, I know. It's just the GCS messages being parsed wrong. note it's spewing off all kinds of numbers too? It just happened to see "aud" in the string. On that FlightData screen in th elower left box, look at the Messages tab, you'll see all the text it's trying to read. 

What version of ArduPilot are you running? That is also read out-loud from that messages menu on connect.

meee1, 2016-10-29:
the words is baud.

38400 baud

or billions of Australian dollars

kikislater, 2016-10-29:
arduplane 3.6.0 and mission planner 1.3.41
I don't have this problem with 1.3.39 and under linux with festival.

It surprises me until I understand the problem