ArduPilot/MissionPlanner #1408: Support for PX4 parameter protocol

LorenzMeier, 2016-10-28:
Some PX4 users trying to use mission planner seem to end up having corrupted parameters due to the different interpretation of the MAVLink parameter field. Would it be possible to either add support for PX4 or to alternatively as a stopgap refuse the connection with an incompatibility message?

I would hate to do the connection refusal on the PX4 side because if MP adds support later that would preclude the use of MissionPlanner even after it has added support.

meee1, 2016-10-31:
Ive modified the display, the issue was around casting the valid value to float, and loosing precision.

https://github.com/ArduPilot/MissionPlanner/commit/93327c0b9755d2ede0706db3d0ac964fec8dc678

LorenzMeier, 2016-10-31:
Thanks, but if my understanding of CS is correct this will not fix it. The slightly unusual choice of using an union for the parameters means that you would need to do this for PX4 (its C++, the equivalent needs to be done for CS):

```C
union {
float f;
uint32_t u;
int32_t i;
} p_union;

p_union.f = mavlink_msg.param_val;

if (type == MAV_PARAM_FLOAT) {
  float param = p_union.f;
} else if (type == MAV_PARAM_INT32) {
  int param = p_union.i;
} else if (type == MAV_PARAM_UINT32) {
  unsigned int param = p_union.u;
}
```

meee1, 2016-10-31:
yep already sorted.
https://github.com/ArduPilot/MissionPlanner/commit/eea9f4f3b4b762257166542961ca26ac155c3783
I updated the method used, but this was already working, before this change.

LorenzMeier, 2016-10-31:
Ok, just checking since I don't want to create ripples for APM users: You are using it differently when talking to APM and you've verified that its still working there?

Thanks!

meee1, 2016-10-31:
https://github.com/ArduPilot/MissionPlanner/blob/aaa4a7ff8fa88fdbee6d6f1f7721b2d74d6e6556/ExtLibs/Mavlink/MAVLinkParam.cs

this is the most direct translation to the union. you will see all data is transported though a byte array, and read as the native type.