ArduPilot/MissionPlanner #1411: Feature Request: Calibrate Airspeed Sensor

Jman841, 2016-11-12:
Could we add a Calibrate Airspeed Sensor button, or the same thing that we have now with doing preflight calibration, however, add a 10 second timer to it?

It is very difficult to properly cover the pitot tube and click the "Yes" to start the airspeed calibration if you are not with an assistant. A simple 10 second timer before it starts the preflight calibration would be hugely helpful.

Thank you

gediminasgu, 2016-11-13:
Just use a marker or a pen cap to cover the pitot :)

Jman841, 2016-11-14:
I have had issues with putting something over the pitot tube where it will block the static ports and not give me a good calibration. A simple 10 second timer can allow me to use my hands to ensure I am not touching the pitot tube.

Great suggestion though! Thank you!

magicrub, 2016-11-14:
Yeah, I think the pen cap thing is the way to go here. I use a 3D printed
thing.