ArduPilot/MissionPlanner #1412: Missing Parameter Descriptions

Code-Tester, 2016-11-14:
I can't see my Parameter Descriptions and Units in the Mission Planner when i add new Parameter in the ArduCopter Firmware. I have tried to modify the ParameterMetaDataBackup.xml File without success. 

Normally it should work, right?

What i have to do? Is it necessary to recompile the MissionPlanner with new Descriptions and Units?

I use the MissionPlanner Version 1.3.41

magicrub, 2016-11-14:
Nope, that text is pulled from the internet from the latest release. Custom params will show the string and value but not description.

Code-Tester, 2016-11-14:
Is there a way to change this? I mean the source for the "Descriptions", perhaps it will be better to have an own Parameter-File offline for the MissionPlanner. So everybody can modify his own file.

meee1, 2016-11-21:
edit ParameterMetaDataBackup.xml in my documents > mission planner.

Code-Tester, 2016-11-21:
Thanks. This works for me.

closed 2016-11-21 by Code-Tester

saintofinternet, 2020-08-30:
> 
> 
> edit ParameterMetaDataBackup.xml in my documents > mission planner.

what should be edited in that file?