ArduPilot/MissionPlanner #1413: DO_DIGICAM_CONTROL mission command is sending ShutterCmd and CommandID x 10,000,000

rmackay9, 2016-11-19:
With the arrival of AC3.4, the Do-Digicam-Control mission item is being corrupted when stored/saved to ardupilot.  I've investigated and it seems that when MP sends a mission_item_int (http://mavlink.org/messages/common#MISSION_ITEM_INT) it is multiplying the x and y fields by 10,000,000.  This makes sense for commands in which those fields hold lat/lon but this command uses those fields for different purposes so they shouldn't be multiplied by anything.

The issue is raised and discussed here: http://discuss.ardupilot.org/t/apm-copter-v3-4-2-rc2-the-command-do-digicam-control-does-not-trigger-the-servo-camera/12816

And here's the corresponding ArduPilot issue: https://github.com/ArduPilot/ardupilot/issues/5215

Seidenschwang, 2016-12-06:
There exists already a commit in the File "MissionPlanner/Mavlink/MAVLinkInterface.cs" from Oborne:

["MAVLinkInterface: dont scale x,y for digicam mission items"](https://github.com/ArduPilot/MissionPlanner/commit/442fea05420528d1a0bfd012fa9d30c97202ac48#diff-3d22e8b9c35a62cf3528da7c9a5d5198)

samucs, 2017-01-06:
Hi, is this fixed for Mission Planner 1.3.43?

Seidenschwang, 2017-01-07:
Sorry, the problem still exists. 

When i create a mission with DO_DIGICAM_CONTROL it needs the ShutterCMD with number "1". After writing the mission via USB to the Pixhawk (Write WP's) i clear the mission in the planer and click "Read WPs". Then i receive the ShutterCMD with "1E-07" :-(

Here the screenshots:
**Before upload:**
[image]

**After download:**
[image]

**Used versions:**
MissionPlanner 1.3.43.2
ArduCopter V3.4.2

meee1, 2017-01-07:
@Seidenschwang  thanks for the report, I can confirm its a display bug, it saves it as 1, but displays it wrong. I will post a fix very soon

rmackay9, 2024-04-22:
this is long since fixed I think.

closed 2024-04-22 by rmackay9