ArduPilot/MissionPlanner #1414: 4k screen resolution issue

buckker, 2016-11-20:
Hi

I have a problem with mission planner and my UHD screen (3840x2160). If I open the program everything looks nice, but after connecting to the pixhawk the program window is much smaller (1/4 of the normal screen size) If I maximize the window to the full screen it's impossible to read something. Everything looks crazy...

I'm not sure if it's possible to modify the manifest file to get rid of this problem... Unfortunately I'm not the specialist for that type of problems.. :-)

Thanks a lot

Michael

meee1, 2016-11-21:
can you please try the latest beta MP.

help > update beta

buckker, 2016-11-22:
I have tested it a few minutes ago. It works :-) Everything looks fine. Thanks for solving this issue!!

popwebz, 2016-11-23:
I also noticed this bug after updating from 1.3.40 to 1.3.41 on my retina MacBook running Windows 10. It appears that it was fixed in the latest beta though.