ArduPilot/MissionPlanner #1415: Add proximity viewer

rmackay9, 2016-11-25:
AC3.5 will include sending proximity information to the ground station using the DISTANCE_SENSOR mavlink message.  In total it will send 8 distances (orientation parameter will be from 0 ~ 7). http://mavlink.org/messages/common#DISTANCE_SENSOR

It would be nice to have an optional viewer window in the MP so that people can easily see the distances seen by the proximity sensor.  Below is just my idea of what it might look like.  It would be good to have both the distance written as a number and expressed as a bar (or something).

[image]

DonLakeFlyer, 2016-11-25:
What's the timeframe for this. Just trying to fit in QGC release timeline.

DonLakeFlyer, 2016-11-25:
These MAV_SENSOR_ORIENTATION style roll/pitch/yaw enums are just awful. When someone adds a new orientation you need a new version of a ground station. Can't the roll/pitch/yaw just be real values in the message?

rmackay9, 2017-01-10:
I'm going to close this issue because it's working in MP now.  There are some improvements that still need making but the basics are there.

@DonLakeFlyer, sorry, I missed your question originally.  This will go out with AC3.5 which enters beta testing within 2 weeks.  I would expect the official release somewhere near the end of Feb is all goes well (but does it ever :-)).
I don't have an immediate answer for the enums.  The messages are there and they're a half-decent (but not fully decent) starting point I think.
It's definitely true that longer-term, we need a more sophisticated way of getting the companion computer's "local barrier" information into ardupilot.  That might involve sending in a mini-fence which is in "earth frame".  We haven't clarified how we're going to do that but that might allow a better set of info to be sent to the GCS.

closed 2017-01-10 by rmackay9