ArduPilot/MissionPlanner #1417: Error when more than one parameter changed

night-ghost, 2016-11-28:
[image]

meee1, 2016-11-28:
can you tell me which parameters? ive been trying to track this down.

night-ghost, 2016-11-28:
any 2 or more parameters. 

fail here 
```

temp.Sort((a,b)=>
            {
                if (a == null || b == null) return 0;
                if (a.EndsWith("ENABLE")) return 1; return -1;
            });
```

night-ghost, 2016-11-28:
may be it will be better not to sort but instead make a parameters loop twice?

meee1, 2016-11-28:
the code has changed since the version you are using. I was trying to discover the cause while under the debugger. I'm pretty sure its fixed in the latest beta.

night-ghost, 2016-11-28:
this sort function not uses 'b' value at all so it REALLY is not consistent

night-ghost, 2016-11-28:
it should return 0 if both params with ENABLE or witout ENABLE

night-ghost, 2016-11-28:
just updated sources from Github - code is the same

```
temp.Sort((a,b)=>
            {
                if (a == null || b == null) return 0;
                if (a.EndsWith("ENABLE")) return 1; return -1;
            });
```

meee1, 2016-11-28:
i duno where your looking.

        static int sorter(string a, string b)
        {
            if (a == null || b == null) return 0;
            if (a.EndsWith("ENABLE") && b.EndsWith("ENABLE")) return a.CompareTo(b);
            if (a.EndsWith("ENABLE")) return -1;
            if (b.EndsWith("ENABLE")) return 1;
            return a.CompareTo(b);
        }

night-ghost, 2016-11-28:
Hmm sorry Git refuses to update. Looks much better and seems that should to work, but...

night-ghost, 2016-11-28:
just tested 1.3.41 - the same error :(

meee1, 2016-11-28:
you need MP beta
help > update beta.

night-ghost, 2016-11-28:
Ok thanks, Beta works fine! This error prevents me form [debugging this](https://www.rcgroups.com/forums/showthread.php?t=2784349) - EEPROM emulated in Flash so all parameters erased on each firmware upload and should be reloaded from file

magicrub, 2016-11-28:
wouldn't it be better to check for EndsWith("_ENABLE") with the _ in there?

meee1, 2016-12-03:
thanks for the report.

closed 2016-12-03 by meee1