ArduPilot/MissionPlanner #1418: Mavlink decoding race condition?

mredstall, 2016-11-30:
We experience an intermittent issue where mission planner reports a disarm/arm cycle during flight (the flight controller is unaffected, this is a mission planner _reporting_ issue). The behaviour is present but not deterministic during log replays.

Looking at the code which updates armed status via heartbeat messages, at break-point the issue seems to be illustrated in the screenshots below. the contents of `hb` are all zero despite the assignment from `mavLinkMessage` which has non-zero values.

[image]
[image]

meee1, 2016-12-01:
I would be interesting to see the contents of mavlinkmessage.buffer, under the bad state.

just trying to determine what part is failing, I assume the structure conversion. but I could be wrong.

any chance of capturing that?

mredstall, 2016-12-01:
Is this what you were after?

[image]

meee1, 2016-12-01:
yes and no.

to determine whats actually failing really need to see if its the object creation that's causing the issue.

I have a feeling mavutils.ByteArrayToStructure is failing, but I want to know whats going into that function for it to fail. its in charge of converting the byte array to the structure. and could explain the results. but I still would like to know the cause. so I can address that

how are you creating this scenario? maybe I can recreate?

mredstall, 2016-12-01:
I am creating the scenario through log playback with a log that has confirmed instances of the issue occurring (it was first evident during live flight, this is what brought our attention to it). During the period between 60-80% of the way through the log file play back it occurs (at different points) at least once more often than not. I could send you through the log file for you to use if you would like?

magicrub, 2016-12-01:
Yes, please link us to the log.

mredstall, 2016-12-01:
https://www.dropbox.com/s/gx038l8ku1m09qk/2016-11-11%2015-45-07.tlog?dl=0

jmachuca77, 2016-12-01:
I have experienced this same issue on several flights.

meee1, 2016-12-02:
@mredstall can you test the latest changes. I think its fixed.

ive added a lock around the entire packet struct creation, as I believe the null check was returning a created blank packet.

mredstall, 2016-12-05:
Issue appears to be fixed. I have played through the log file several times and have not been able to recreate the issue.

meee1, 2016-12-05:
@mredstall thanks for reporting back.

closed 2016-12-05 by meee1