ArduPilot/MissionPlanner #1419: MP does not take input at FlightData screen when double-clicked on Tuning window until Graph This page shows which takes 30s

ferdicakmak, 2016-12-01:
Hello,

When I connect ArduPlane V3.7.1 on Pixhawk to MissionPlanner 1.3.41.8 and activate “tuning” screen everything works fine and realtime data flows smoothly.

When I double-click on “tuning” screen, “ComPort”, “BaudRate” comboboxes and “Connect/Disconnect” button start to flicker like changing focus, “connect picture” changes between green and red (connected, disconnected) irregularly and it takes about 30 seconds for “Graph This” screen shows up. And also the strings in the disabled connection comboboxes are deleted sometimes (Restored when MP is resized.).

[image]

In this 30s interval, MP1.3.41.8 does not take any inputs. When “Graph This” window shows up, everything goes back to normal.

I didnot have problems with this when using earlier versions. And i get the same problem with my powerfull (i7-processor) laptop and a weak (celeron) laptop. I can continue operation with powerfull machine after 30s, but it becomes impossible to go on with a weak computer.

Any suggestions? 

Thanks,
Ferit.

meee1, 2016-12-01:
Thanks for the report. Ive found the issue, and will push a new beta soon

meee1, 2016-12-03:
fixed thanks

closed 2016-12-03 by meee1

ferdicakmak, 2016-12-03:
Thank you for your quick response and action.

Best regards.
Ferit.