ArduPilot/MissionPlanner #1425: bug:  cam_trig_type and rc_function will set back to wrong values after open "gimbal setup"

SaschaHelle, 2016-12-09:
i folks, 
after while of testing and some help from chris jackson and philip rowse i figure out the problem, why my camera triger in pixhawk release 3.4.3 not work.. 

following steps i do :
change cam_trig_type from "relay to servo" 

change on ( in my example ) rc12 _function from  1- pass through to 10 - camera trigger.
save and write params, 
check trigger, everything works fine .

after that i open camera gimbal page, 

don´t change any values, don enter in any fileds, just open, and leave the page. 

after that, the values "cam_trig_type  is set back to "relay" 
and funkction on rc12 is set back to "disable" 

trigger will not work. 

in my opinion, that is a bug and should fixed..

thanks for doing that !

regards sascha

aloerch, 2017-08-08:
This is still a problem that many people are encountering. I'm commenting here, even though this post is a bit old, because the behavior of MP is the same in the current version. What I've found is that in the Gimbal Setup page, there is an option for Camera Shutter, and there is a drop down field to select the camera shutter type. Changing the parameter for CAM_TRIG_TYPE in MP's full parameter list does NOT update the drop down field for the Camera Shutter Type in the Gimbal Setup page. While that could be considered an annoying problem, what's worse is that when you update the parameter for CAM_TRIG_TYPE in the full parameter list, you must "write params" to save them.... but when you open the Gimbal Setup, the parameters are changed and written to the autopilot with no confirmation. This is why the CAM_TRIG_TYPE keeps changing everytime you (or almost anyone else) visits the Gimbal Setup page. 

A workaround for this is to go into Gimbal Setup, and change the Camera Shutter type to what you want it to be (Relay, Servo #, etc). As long as you make sure your settings in the Gimbal Setup page match what you want set in the full parameter list, this will prevent your camera trig type from changing unexpectedly.

A solution for the MP developers would be to either 1) Implement a save settings confirmation on pages like the Gimbal Setup to prevent unintended parameter changes (why force a full parameter list confirmation button "write params" but not a confirmation for other pages that change parameters, including customized ones) and 2) When a user changes a parameter in the full parameter list, ensure that change is reflected in the GUI pages to prevent a mis-match between what a user wants a parameter to be, and what it winds up being.

aloerch, 2017-11-06:
Bump.... this needs attention.