ArduPilot/MissionPlanner #1426: Exporting BIN files to Dashware compatible format

lvale, 2016-12-10:
Hi
The Dashware app is used to overlay data files from various telemetry protocols on flight footage and there's several approaches to have it done like https://www.youtube.com/watch?v=Qd9irkB-lk4 

Can I suggest an additional export function to a DashWare compatible format ?

magicrub, 2016-12-10:
Sounds like we just need to make an import filter for MAVLink.

lvale, 2016-12-10:
@magicrub  ?? Import filter ?? I believe it's just a matter of exporting fields on a csv file ordered by columns....

lucamax, 2017-11-30:
It would be great, since Dashware is was bought by GoPro and is no longer developed but it is free. 
The actual Mission Planner BIN to CSV function has the wrong format so to do it you must as far as I know use an utility https://github.com/OverloadUT/apm-dashware and then  filter the data with  Excel , Libre Office will not work since files are easily about 12 mb. A nightmare

meee1, 2017-11-30:
is there a spec on the format required?

lucamax, 2017-11-30:
_Here are some notes on making a DashWare-friendly CSV data file:

-It’s good to have a static line of text at the top declaring that it is your data format. Something like “Some Cool Logger Data File”. The DashWare Data Profile for your format can then be easily set up to automatically detect one of your data files.
-We require a CSV header line: A line of text containing the column headers/names.
-Then the raw data lines would follow after that.
-There should be a Time column (full timestamps work great, e.g. “2011/08/07 20:10:02.200”, or the total running time in seconds e.g. “1.1”)
-Latitude and Longitude should be in +/- decimal degrees
-Be consistent with your data units, or at least change the column name when you use different units. DashWare can be used to create calculated data columns for other units.

You will then need to make a Data Profile that understands your format. Bring up the Data Profile Editor (File Menu > Edit Data Profiles…) and press F1 for detailed help. Tips:
-The column mappings (bottom right area of dialog) are used to map your column names to known DashWare data types. Set up the mapping for your time column first, so you can use the “Test File / View Data…” button. It is very handy. “ / Data Running Time, Seconds” is the mapping you need.
-To get a map to show up, you need to use our “Degrees to Cartesian Converter” (calculators/converters are in the bottom left area of dialog) to create X and Y units in feet. You must map these new X and Y output columns to our “Positional \ Longitude / X Position (Feet)” and “Positional \ Latitude / Y Position (Feet)” data types._

lucamax, 2017-11-30:
Some users made incredible work with Dashware   , i.e https://youtu.be/zv_6BY_IzN0?t=74

lvale, 2017-11-30:
@meee1 Example here

https://github.com/gulp79/apm-dashware

meee1, 2017-11-30:
ive added the csv file creation in the control-f screen

https://github.com/ArduPilot/MissionPlanner/commit/fcfd1d02e258db8dea5f99f984bed43e2868a57b#diff-91a0546f117aabdbe5f04c2cad1c6d4cR1

lvale, 2017-11-30:
Thanks Michael @meee1 

Closing

closed 2017-11-30 by lvale